theCore-embedded/theCore #19: Add error handling to i2c bus implementaion

vadimol, 2016-04-25:
For now i2c bus implementation does not support error handling. Actually device can hang in case of en error or glitch on a bus.
For polling we can use timeouts for operations.
For irq  - there is a separate handler for errors.

rasendubi, 2017-05-10:
We should also switch to using `I2C_GetLastEvent` instead of `I2C_GetFlagStatus` because the former allows checking errors.

forGGe, 2017-05-10:
@rasendubi Make sense.

forGGe, 2017-07-10:
@Dubland : check https://github.com/forGGe/theCore/issues/19#issuecomment-300573650

forGGe, 2017-07-10:
Function usage is described in [stm32 i2c source](https://github.com/forGGe/spl-hub/blob/master/STM32F4xx_StdPeriph_Driver/src/stm32f4xx_i2c.c#L955-L1053).

Checks like these https://github.com/forGGe/theCore/blob/develop/platform/stm32/export/aux/i2c_bus.hpp#L586 must be re-implemented using `I2C_GetFlagStatus`. 

Error can be reported using bus event:
https://github.com/forGGe/theCore/blob/develop/platform/common/export/common/bus.hpp#L26

After that, `soft_reset` can try to repeat xfer if necessary:
https://github.com/forGGe/theCore/blob/develop/dev/sensor/htu21d/export/dev/sensor/htu21d.hpp#L177

Talk to @vadimol for more details of how i2c driver works if something unclear.

rasendubi, 2017-07-10:
A small correction: checks must be re-implemented using `I2C_GetLastEvent`.

`I2C_GetFlagStatus` resets some status flags so is not appropriate to call multiple times in a row.

forGGe, 2017-10-11:
Implemented as a part of #279 Thanks to @Dubland !

closed 2017-10-11 by forGGe