theCore-embedded/theCore #234: Use CMSIS-compliant interrupt names

forGGe, 2017-02-20:
The startup assembly files [uses reference to internal handler](https://github.com/forGGe/theCore/blob/develop/arch/arm_cm/startup_arm_cm.S#L94). Instead, device-specific CMSIS-compliant names must be used.

rasendubi, 2017-05-10:
@forGGe mark "easy task"?

forGGe, 2017-05-10:
@rasendubi Not really - scope is somewhat large. It includes:
 - changing ach-level assembly, so that every platform will supply own list of interrupts (vendor-specific). [Draft for ke02](https://github.com/forGGe/theCore/commit/ecc2a56cb28b9c5bfe16f502fd5071bfd6c8d1b9) illustrates it.
 - adding device IRQ assembly list file per each ARM-based platform.
 - dropping usage of [the irq manager](https://github.com/forGGe/theCore/blob/develop/platform/common/irq.cpp) in every platform driver, since it is useless in case when every IRQ has its own symbol. Platform drivers should rely on new IRQ symbols. 
 - buildsystem change and preprocessor stuff - every device instance must be noted in top-level project cmake (one that will use theCore as library) file, so that proper IRQ handlers will be created instead of weak symbols noted in assembly.

forGGe, 2017-06-16:
Related to the #227