theCore-embedded/theCore #247: Drop clock routine usage from wait_for function (and make systimer mandatory)

forGGe, 2017-04-04:
Systimer implementation: #246 

Description to be filled.

vadimol, 2017-06-17:
1) Each platform provides `platform_spin_wait()` and (optionally) `SYSTMR` implementation.
2) Common-execution provides `spin_wait() `and `wait_for()`, The `wait_for()` uses ` spin_wait()` in case `SYSTMR` is not available.
3) The thread-lib provides `sleep_for()` that uses `wait_for()` from common-execution or OS specific interface if available.
@forGGe , what do you think?

forGGe, 2017-06-17:
@vadimol looks good but have one question. What is relationship between `spin_wait()` and `platform_spin_wait()` ? If `spin_wait()` will call `platform_spin_wait()` no matter what, then it is reasonable to keep `spin_wait()` only in platform headers (they are exposed via `common/execution.hpp` in any case)

forGGe, 2017-07-04:
@vadimol thanks! implementation completed, closing this issue.

closed 2017-07-04 by forGGe