theCore-embedded/theCore #256: HTU21D example does not work in release mode

forGGe, 2017-05-15:
See analyser output:

[image]

rasendubi, 2017-05-15:
I'll decipher what happens:
- The first two types issue a soft reset command, so device goes down.
- The third and last byte is a start of a temperature read command.

It fails because the device did not answer with an ACK (due to it being resetting). Now the I2C driver is waiting for a Master Transmitter Mode Selected event, which never occurs due to the error.

This issue directly relates to #19. I would say, this issue is an instance of absent error handling.

A fast workaround is to add a ~15ms delay to allow HTU21D complete reset before issuing the read command.

forGGe, 2017-05-15:
> A fast workaround is to add a ~15ms delay to allow HTU21D complete reset before issuing the read command.

@Dubland, since you are working on #183 , it will be useful to you also check @rasendubi's workaround.

rasendubi, 2017-06-14:
ping @Dubland ?

Dubland, 2017-07-01:
With delay after soft reset example working fine. Shall I create PR for adding the delay?

rasendubi, 2017-07-02:
> Shall I create PR for adding the delay?

Yes please.

forGGe, 2017-07-02:
@Dubland now, when #271 is merged, I propose to try find real root cause and fix it. @rasendubi can guide you trough it.

forGGe, 2017-08-01:
@Dubland any news?

forGGe, 2017-10-11:
Implement in #279 Thanks to @Dubland !

closed 2017-10-11 by forGGe